FSEC/SPD_server #17: Evaluate allowing 2 modules per microinverter

kshowalter, 2018-04-17:


mechkit, 2018-05-10:
"mppt_channels" is in the database for inverters, and serves no other use on microinverters.
This seems like a good variable to use to designate a double inverter. 
Example: https://usa.apsystems.com/wp-content/uploads/2018/03/APsystems-Product-Datasheet-YC600-8.31.17.pdf

mechkit, 2018-05-15:
String size selection algorithm update: https://github.com/FSEC/SPD_server/commit/c67025d2bce7a04c4ac2742dce59fb29ed6306bd

mechkit, 2018-05-16:
It seems like all the system checks will work with minor changes (https://github.com/FSEC/SPD_server/commit/1ef40e4f2416ffa270214f2d2dc1b83821f16f04). Testing should show if anything is missed.

mechkit, 2018-05-22:
Updated drawing. https://github.com/FSEC/SPD_server/commit/c0e31d6acdb359aaa07d733d310eb455efaaff98

mechkit, 2018-05-22:
Updated drawing inverter totals on table. https://github.com/FSEC/SPD_server/commit/865279a82b82585d803d6bdb7144fb6ecbea2a60

mechkit, 2018-05-22:
Array totals and wire sizing seems to handle the transition from on tracker to two.

mechkit, 2018-05-22:
Since the code work has been done, I would say the evaluation is that this can and should be done.

Closing this issue, and continuing the work in https://github.com/FSEC/SPD_server/issues/32

closed 2018-05-22 by mechkit