FSEC/SPD_server #25: Some PDFs are being corrupted/locked on the network share

kshowalter, 2018-04-27:


kshowalter, 2018-04-27:
This file was locked and had zero size, causing the failure of certificate generation.
\\shares.fsec.ucf.edu\spd\Drawings\PV_drawing_2201.pdf

kshowalter, 2018-04-27:
The above file was deleted, and the SPD generated a new file for the SCP certification.

kshowalter, 2018-04-27:
~2018-04-26
A customer is having a difficult time downloading the certificate for request 700230.  It looks like the same issue that affected 700220. The drawing for this project, PV_drawing_2356.pdf, is another 0 byte file that is locked for being "in use" in the drawings folder.

kshowalter, 2018-04-27:
Above system (2356) pdf can not be deleted until SPD server is rebooted. Temporary PDF files do not seem to be affected, only the final merged PDF created by Ghostscript.

kshowalter, 2018-04-27:
https://github.com/FSEC/SPD_server/commit/354418b1e87b201185b7edf035ea96e4aaf1eb16 changes the ghostscript command to write the file locally (to the SPD server), then adds a command to the OS to copy the file to the network share.

mechkit, 2018-05-31:
No further problems. Closing this for now.

closed 2018-05-31 by mechkit